jeanthom/Otter-Iron-PRO-BOM #3: Fix diode part number

jeanthom, 2020-06-28:
The chosen P/N for the diode is a THT component. Vishay B140-E3/61T should be the right component to use.

jeanthom, 2020-07-10:
This diode is too big for the PCB.

ElRoberto538, 2020-09-13:
Hi, thanks for making the parts list. Did you manage to figure out which diode to use?

jeanthom, 2020-09-13:
Nope, I went with the B140-E3/61T anyway, scrapping the soldermask to make it fit.

jeanthom, 2020-09-13:
@Jan--Henrik What P/N did you use?

ElRoberto538, 2020-09-13:
@jeanthom Found it, it is listed in the BOM csv for the normal Otter Iron, the part number is listed incorrectly but the LCSC part number gives the correct diode.

https://lcsc.com/product-detail/Schottky-Barrier-Diodes-SBD_MDD-Microdiode-Electronics-DSK210-K210_C64891.html

jeanthom, 2020-09-13:
Fixed in git repo, thank for your help

closed 2020-09-13 by jeanthom