jeanthom/Otter-Iron-PRO-BOM #7: Incorrect USB interface controller

ElRoberto538, 2020-12-13:
Hi, I've been debugging my otter Iron for the last couple of days, and after testing everything I can think of and customizing the firmware to try and debug the USB PD profile, I finally worked out that I have the wrong STUSB4500 controller! The "L" version apparently doesn't support USB PD... The link below is for (what I believe) is the correct chip;

https://www.arrow.com/en/products/stusb4500qtr/stmicroelectronics

jeanthom, 2020-12-13:
You're damn right. I ordered the non-L version by chance because both L/non-L versions weren't available from LCSC and Arrow only had the QTR version in stock.

Let me know if you have difficulties sourcing the non-L version.

ElRoberto538, 2020-12-13:
Thanks, I found them in stock at Element 14, along with the current sense resistor I need, fingers crossed they arrive before Christmas break!

ZipperZ, 2021-01-01:
Oh... got on the same "hook". LCSC had the "L" in stock... Oh well

ElRoberto538, 2021-04-07:
I think I have a few spare, unfortunately unless you're in New Zealand the shipping is likely to be either very pricey or very slow (or both!).

cheesemarathon, 2021-04-08:
> I think I have a few spare, unfortunately unless you're in New Zealand the shipping is likely to be either very pricey or very slow (or both!).

Thanks, but I'm in the UK. I'm looking at buying a https://shop.pimoroni.com/products/sparkfun-power-delivery-board-usb-c-qwiic as that appears to have the correct chip on it to remove and put on my Otter Iron!